priv-kweihmann/oelint-adv #722: class-nativesdk from include in bbappends isn't known from PACKAGES, MACHINE, DISTRO

skycaptain, 2025-02-28:
I have a .bbappend file which includes an .inc file similar to 

`cmake_%.bbappend`:

```bitbake
require ${@bb.utils.contains('DISTRO_FEATURES', 'my-feature', '${BPN}_my_feature.inc', '', d)}
```

`cmake_my_feature.inc`:

```bitbake
do_install:append:class-nativesdk() {
    ...
}
```

I get the following error from oelint-adv 7.2.1:

```
meta/recipes-devtools/cmake/cmake_my_feature.inc:17:error:oelint.func.specific:'do_install' is set specific to ['class-nativesdk'], but isn't known from PACKAGES, MACHINE, DISTRO [branch:true]
```

I'm not sure what the issue is here. It might be that I haven't checked out poky which contains the original recipe, thus oelint might not know that it defines `BBCLASSEXTEND = "nativesdk"`

priv-kweihmann, 2025-03-01:
Hi, I tried to reproduce the issue, I don't get any warning.
Are you sure your local environment has everything installed in the latest version?
Recently `oelint-data` was added as a dependency, which needs to be up to date

skycaptain, 2025-03-02:
I've created a MRE here: https://github.com/skycaptain/oelint-adv-issue-722 Running oelint-adv either standalone or as pre-commit hook results in the following:

```shell
$ pre-commit run --verbose --all-files oelint-adv
Advanced oelint..........................................................Failed
- hook id: oelint-adv
- duration: 0.41s
- exit code: 1

Loaded rules:
        oelint.append.protvars
        oelint.append.protvars.LICENSE
        oelint.append.protvars.LIC_FILES_CHKSUM
        oelint.append.protvars.PR
        oelint.append.protvars.PV
        oelint.append.protvars.SRCREV
        oelint.bbclass.underscores
        oelint.exportfunction.dash
        oelint.file.inactiveupstreamdetails
        oelint.file.inappropriatemsg
        oelint.file.includenotfound
        oelint.file.includerelpath
        oelint.file.inlinesuppress_na
        oelint.file.nospaces
        oelint.file.patchsignedoff
        oelint.file.requireinclude
        oelint.file.requirenotfound
        oelint.file.underscores
        oelint.file.upstreamstatus
        oelint.func.specific
        oelint.newline.consecutive
        oelint.newline.eof
        oelint.spaces.emptyline
        oelint.spaces.linebeginning
        oelint.spaces.linecont
        oelint.spaces.lineend
        oelint.tabs.notabs
        oelint.task.addnotaskbody
        oelint.task.customorder
        oelint.task.dash
        oelint.task.docstrings
        oelint.task.heredocs
        oelint.task.multifragments
        oelint.task.network
        oelint.task.noanonpython
        oelint.task.nocopy
        oelint.task.nomkdir
        oelint.task.nopythonprefix
        oelint.task.order
        oelint.task.order.do_build
        oelint.task.order.do_compile
        oelint.task.order.do_configure
        oelint.task.order.do_fetch
        oelint.task.order.do_install
        oelint.task.order.do_package
        oelint.task.order.do_patch
        oelint.task.order.do_populate_sysroot
        oelint.task.order.do_unpack
        oelint.task.pythonprefix
        oelint.var.addpylib
        oelint.var.badimagefeature
        oelint.var.badimagefeature.allow-empty-password
        oelint.var.badimagefeature.allow-root-login
        oelint.var.badimagefeature.dbg-pkgs
        oelint.var.badimagefeature.debug-tweaks
        oelint.var.badimagefeature.dev-pkgs
        oelint.var.badimagefeature.eclipse-debug
        oelint.var.badimagefeature.empty-root-password
        oelint.var.badimagefeature.post-install-logging
        oelint.var.badimagefeature.ptest-pkgs
[... 218 lines cut ...]
        oelint.vars.pkgspecific.pkg_prerm
        oelint.vars.pnbpnusage
        oelint.vars.pnusagediscouraged
        oelint.vars.pythonpnusage
        oelint.vars.renamed
        oelint.vars.sectionlowercase
        oelint.vars.spacesassignment
        oelint.vars.specific
        oelint.vars.srcuriappend
        oelint.vars.srcurichecksum
        oelint.vars.srcuridomains
        oelint.vars.srcurifile
        oelint.vars.srcurigittag
        oelint.vars.srcurioptions
        oelint.vars.srcurisrcrevtag
        oelint.vars.summary80chars
        oelint.vars.summarylinebreaks
        oelint.vars.unpackdir
        oelint.vars.valuequoted
        oelint.vars.virtual
Parsing [redacted]/meta-oelint-adv-issue-722/conf/layer.conf
Parsing [redacted]/meta-oelint-adv-issue-722/conf/layer.conf
Parsing [redacted]/meta-oelint-adv-issue-722/recipes-devtools/cmake/cmake_%.bbappend
meta-oelint-adv-issue-722/recipes-devtools/cmake/cmake_%.bbappend:1:error:oelint.func.specific:'do_install' is set specific to ['class-nativesdk'], but isn't known from PACKAGES, MACHINE, DISTRO [branch:true]
```

closed 2025-03-03 by priv-kweihmann

skycaptain, 2025-03-03:
Can confirm 7.2.3 fixed this issue as well

commit 037f3a15 (2025-03-03): core: enforce fork as mp context

as we need to share all resources of the linter's
main process with the child tasks running the checks.
Otherwise things like the CONSTANTS interface will
be simply reinitialized within the child's space,
but ignoring all the added things from the tweaks
runs by the main process.

This will be mandatory from python 3.14

Closes #721
Closes #722
commit 26262f88 (2025-03-03): core: enforce fork as mp context

as we need to share all resources of the linter's
main process with the child tasks running the checks.
Otherwise things like the CONSTANTS interface will
be simply reinitialized within the child's space,
but ignoring all the added things from the tweaks
runs by the main process.

This will be mandatory from python 3.14

Closes #721
Closes #722
commit 222039f1 (2025-03-03): core: enforce fork as mp context

as we need to share all resources of the linter's
main process with the child tasks running the checks.
Otherwise things like the CONSTANTS interface will
be simply reinitialized within the child's space,
but ignoring all the added things from the tweaks
runs by the main process.

This will be mandatory from python 3.14

Closes #721
Closes #722
PR #724 (2025-03-03): core: enforce fork as mp context
as we need to share all resources of the linter's
main process with the child tasks running the checks. Otherwise things like the CONSTANTS interface will be simply reinitialized within the child's space,
but ignoring all the added things from the tweaks
runs by the main process.

This will be mandatory from python 3.14

Closes #721
Closes #722