priv-kweihmann/oelint-adv #744: var.multiinherit and and vars.duplicate not behaving as expected

stwalser, 2025-05-02:
I noticed another interesting behavior regarding .bbclass files.

In some, but not all of my .bbclass files, multiinherit reports that a class is imported multiple times, although the files contain only one inherit statement. In two cases I inherit core-image and in the other two own custom classes.

Additionally, vars.duplicate reports that I have duplicate DEPENDS entries, even though no entry is set twice in the class. Furthermore, the entry is also not set in any parent class. Some of the files this happens for overlap with the files mentioned in the multiinherit context.

priv-kweihmann, 2025-05-02:
A minimal example would be highly appreciated - otherwise it could be anything

priv-kweihmann, 2025-05-16:
@stwalser could you please provide a minimal example how to reproduce the reported issue.
If nothing is provided I will have to close this issue

stwalser, 2025-05-16:
Regarding number 3:

I have two files affected by the issue. The first one has ~30 functions with the following naming scheme:
lorem_ipsum_dolor():
lorem_ipsum_sit():
lorem_ipsum_amet():
...

The first two words are always the same. What comes after is different from all other functions.

In the second file, the naming looks like this:
takimata_sanctus():
takimata_est():
takimata_gubergren():
...

Here only the first word is always the same and what comes after is different.

The functions are independent and cannot be merged. However, I get this output:
oelint.task.multifragments:Multiple fragments of the same function in the same file should be merged [branch:false]

I hope the content of the functions is not relevant.

stwalser, 2025-05-16:
Regarding number 2:

I have a .bbclass file with a DEPENDS statement like this at the top

DEPENDS += "\
    depend1 \
    depend2 \
"

The class does not inherit from any other class. It is at the top of the hierarchy. 
Still I get the following warnings:
oelint.vars.duplicate:Item 'depend1' was added multiple times to DEPENDS [branch:false]
oelint.vars.duplicate:Item 'depend2' was added multiple times to DEPENDS [branch:false]

stwalser, 2025-05-16:
Regarding number 1:

I have a .bbclass file with

inherit core-image

at the top. The file then only contains two functions.
However, I get the following warning:
oelint.var.multiinherit:'core-image' is included multiple times [branch:false]

A thought I had about this, is, that maybe it is inherited again in a class inheriting from this one and it is not reported with correct file.

priv-kweihmann, 2025-05-17:
With

`test.bb`

```
inherit test
```

and `test.bbclass`

```
inherit core-image

DEPENDS += "\
    test1 \
    test2 \
"

lorem_ipsum_dolor() {
    :
}

lorem_ipsum_sit() {
    :
}

lorem_ipsum_amet() {
    :
}

def takimata_sanctus():
    pass

def takimata_est():
    pass

def takimata_gubergren():
    pass

```

I don't see any of the reported issues on oelint-adv `8.0.1`.

stwalser, 2025-05-19:
I tried it with the latest version too and I still have the same issues unfortunately. Presumably, it is an affect due to other files in the project. The problem is just, that my project at hand contains hundreds of .bb files and I need to do a bit more debugging on which files could be contributing to these issues. I will let you know as soon as I know more.

stwalser, 2025-05-20:
I managed to resolve the problem. As suspected, the involvement of other files was key to the problem. I had a recipe that inherited from class B and from class A. However, class B also inherited from class A. As a result, the functions and the DEPENDS statement of class A must've been parsed twice and been reported as multi-fragments and duplicates.

closed 2025-05-20 by stwalser