priv-kweihmann/oelint-adv #764: Conditionally skip checking inherited files?

macpijan, 2025-06-12:
It may be a weird use case, but we have a situation, where we have some files we would not want to adjust to meet the needs of the rules, while most of them should.

We do that via `exclude` in `pre-commit` and it worked great, until we faced to the following case, that one of the excluded files is inherted by the non-excluded one.

Example execution:

```
oelint-adv --rulefile=.oelint-ruleset.json --hide=info --fix --constantmods=+.oelint-custom-overrides.json meta-zarhus-distro/recipes-extended/evsieve/evsieve_1.4.0.bb
```

Example output shows that more files are checked:

```
Parsing /home/macpijan/projects/zarhus/meta-zarhus/meta-zarhus-distro/recipes-extended/evsieve/evsieve_1.4.0.bb
Parsing /home/macpijan/projects/zarhus/meta-zarhus/meta-zarhus-distro/classes-recipe/cargo.bbclass
Parsing /home/macpijan/projects/zarhus/meta-zarhus/meta-zarhus-distro/classes-recipe/cargo_common.bbclass
Parsing /home/macpijan/projects/zarhus/meta-zarhus/meta-zarhus-distro/classes-recipe/rust-common.bbclass
Parsing /home/macpijan/projects/zarhus/meta-zarhus/meta-zarhus-distro/classes-recipe/rust-target-config.bbclass
```

This file inherits cargo, which we wanted to exclude from checks - we assume this is the reason why cargo and rust classes are picked there:

```
cat meta-zarhus-distro/recipes-extended/evsieve/evsieve_1.4.0.bb
inherit cargo
```

Do you see a way of adding conditional switch of disabling such automatic pulling of inherited files, and do you see a place for it in this project?

priv-kweihmann, 2025-06-12:
See https://github.com/priv-kweihmann/oelint-adv/discussions/598 why in general this ain't a good idea.
From my perspective every compbination of bb/bbappend/bbclass/conf files needs to run error free.

And yes if a recipe/append file inherits a class in the same layer it is taking into account automatically.
As it increases the chances to catch issues in the entire implementation.

So my main questions would be

- why can't the bbclass be fixed to not produce errors?
- how would it beneficial to the linter if you could skip entire files (basically rendering all of the results useless)?
- what is the exact error you are seeing?
- can't inline suppressions be of help here?

priv-kweihmann, 2025-06-20:
Closing due to lack of response - feel free to reopen if the flagged questions are answered

closed 2025-06-20 by priv-kweihmann

macpijan, 2025-06-20:
The question has been asnswered, thank you.

We went with fixing the errors in the included bbclass in the end.