priv-kweihmann/oelint-adv #766: False positive from oelint.var.multiinclude

m-iwanicki, 2025-06-18:
**Description**

oelint-adv warns that `*.inc` file is included multiple times which isn't true.

**Reproduction**

To reproduce:

* create layer with 2 recipes and 1 `.inc` file which is included (`require`) in both of them
* create another layer with bbappends for those 2 recipes and new `.inc` file which is included in those `.bbappends`
* pass all files to `oelint-adv`

**Additional information**

Minimal code to reproduce with both pre-commit and standalone oelint-adv: https://github.com/m-iwanicki/minimal-multiinclude
Possible https://github.com/priv-kweihmann/oelint-adv/issues/285 regression?

I mostly noticed this bug with pre-commit, but also happens when calling `oelint-adv` manually.

m-iwanicki, 2025-06-18:
>  Basically you are injecting the same file twice at different points of the generated script.

> So please explain why this is a big in the linter, rather than in the recipe implementation seen by the linter

I'm including different `.inc` files with different names.

priv-kweihmann, 2025-06-19:
I see - fix incoming

commit 0abd8a59 (2025-06-19): core: fix grouping

by taking the full filename into account.
This will prevent version less files, like images,
being grouped together, when their common file prefix
is the same

Closes #766
closed 2025-06-19 by priv-kweihmann