priv-kweihmann/oelint-adv #788: Add ability to ignore specific files

vj-hansen, 2025-08-29:
There is no way to ignore entire files afaict.

This becomes an issue when:
- Some recipes or includes are vendor-provided and not "meant" to be linted.

priv-kweihmann, 2025-08-29:
https://github.com/priv-kweihmann/oelint-adv/discussions/598 - without a more specific example why, I'm afraid "hygiene" stands above vendor specific mayhem

closed 2025-08-29 by priv-kweihmann

vj-hansen, 2025-09-01:
> So the problem is rather with the `oelint.vars.srcuriappend` rule and I agree that one deserves a bit more of touch up, as it doesn't check if there are any weak defines before the warned about operation

Yes, getting `error: oelint.vars.srcuriappend` on files generated with `bitbake -c update_crate`.

sorry about the bad issue ticket..